SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #1: Firmware update instructions

andre68723, 2023-11-25:
Hi,

thanks for providing this, I've been waiting for this since I installed the inverter.
Can you add some info for already setup inverters? (my IOS App only displays the 
Wifi with "good", no other setting visible). 

I'm not sure if I have to redo the setup from the app again as in the Readme.
If not - 

a) required signal strength for update
b) how long until the automatic? update is executed/rolled out
c) displaying firmware version in the app

Thanks, Andre

mawoka-myblock, 2023-11-25:
You don't notice the firmware update on the inverter. You just need to connect via Bluetooth to the inverter to enable local mode. If that doesn't work, make sure that your app is up-to-date

andre68723, 2023-11-25:
Addendum:

using another Ipad with the App not logged in allows adding via direct connection and enabling local mode.
If the Device is already added via Cloud connection, it wont show up in direct connection dialog

CM000n, 2023-11-25:
> Addendum:
> 
> using another Ipad with the App not logged in allows adding via direct connection and enabling local mode. If the Device is already added via Cloud connection, it wont show up in direct connection dialog

You can also simply log out of your profile in the app and then initiate a direct connection 😉

andre68723, 2023-11-25:
yep, thats the important part. Firmware version is then also in the menu 

Isn't it EZ1 "1.6.0" ( vs. 1.1.6)
[image]

mawoka-myblock, 2023-11-25:
That could be, will check on that

> yep, thats the important part. Firmware version is then also in the menu
> 
> Isn't it EZ1 "1.6.0" ( vs. 1.1.6) <img alt="Bildschirmfoto 2023-11-25 um 10 54 58" width="747" src="https://private-user-images.githubusercontent.com/16468234/285578282-4db785a9-a85e-4773-b04d-9bf58fde52bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5MDY1NTEsIm5iZiI6MTcwMDkwNjI1MSwicGF0aCI6Ii8xNjQ2ODIzNC8yODU1NzgyODItNGRiNzg1YTktYTg1ZS00NzczLWIwNGQtOWJmNThmZGU1MmJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI1VDA5NTczMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3OGY3MjIxODcwZDA3NjhkZTc3MjU0ZjM4NjA2NjFlYTU3OGNhNGRlM2M0NWE2OGI3MDhhYjE4Mjc0NzUzZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wj4nJYh87DeWKRlN5Obd6uQ9nAQh7keQADQO-N-azhM">

1stGate, 2024-01-08:
Hi, 
Back to the inital post. 

I connected local via bluetooth. My inverter EZ1-M has Ver. 1.1.1 and Apsystems say when I connect via App "Your firmware is up to date" but this Ver. does not work with the library.

 howto update the firmware to the required version?
 Where can I get / find that version?

1stGate, 2024-02-07:
Hi again,

I'll reply to my last post 

> howto update the firmware to the required version?
> Where can I get / find that version?

and some cupple weeks ago...

The  EZ1-M firmware version 1.1.1 does not support the APsystems-EZ1-API. You have to upgrade to an higher version.

1. Make sure the inverter is connected to the PV-modules >> and << to the AC power line
2. Make sure the EZ1-M is in direct mode ( disconnect from APsystems cloud ) - connect the inverter via Bluetooth.
3. Connect the EZ1-M also to the Internet e.g. via you WLAN - this is an important step.
4. Check the EZ1-M is online - e.g. find out the local IP and send a ping to the EZ1 inverter - don't trust your app on your   phone !
5. Disconnect both PV-modules from the EZ1-M at least for 1 hour, or better wait for the night. after that connect the PV    again - of course you will need some solar power...  
6. Now drink a lot of tea or coffee wait until an update is shown on the easy power app - don't unplug the inverter - never ! 
   This can take a very long time - e.g. hours of hours.

7. After the update show up in your app, click install. This will start the update to your phone. 
8. if the update on the phone is finished. The update goes further to the EZ1. But in my case this started only after 9:00 am. Until that, you have to wait for !
9. At 9:00 am you can start the transfer to the EZ1-M. Again wait a long time, maybe an half an hour to finish the upgrade.
10. That's it is. Check the new firmware version - you are still connect via Bluetooth.

Well, I'll hope that will help for some case

retoch44, 2024-05-21:
Is it possible that the android app in direct mode does not show the real firmware version? Today I installed an EZ1-M (manufacturing date Aug. 2023 I think) and it displayed V 1.0.0, short after that it displayed a firmware update available and I did the update. After completing the update it still displays V 1.0.0 again and that no update is available. Api is working in home assistant.

1stGate, 2024-05-21:
I have updated 2 devices to version 1.7.0. Both worked as expected. A few days later I connected a new solar battery to an inverter. The solar battery app wanted to upgrade the battery firmware to version 1.59. It did that too, but at the same time it “changed” the firmware in the EZ1-M !!! The battery is NOT from APsystems. I deliberately did not set up an inverter in that mobile app !!! For the app there are only unknown inverter. I don't know where the app got the information about my inverter from and how it connect with this. After this upgrade, version 1.0.0 was displayed in the firmware on the inverter as described by  @retoch44 . Since then, I can't upgrade this EZ1-M. Local mode still works some kind, but behaves very strangely. Some values ​​returned by the device are incorrect. I doubt that it actually version 1.0.0 in the EZ1-M is. 

In the APSystems App, it also shows me that the inverter firmware version 1.0.0 is current. 
Which EZ1-M firmware is up to date?