SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #9: New API Library for handling Errors over Night when the inverter is offline

gerhardwallnoefer, 2024-01-08:
Hello Guys, I saw there was an Update of the APsystems EZ1 - Python Library.
What do I have to do to become this changes to my Homassistant Instance?
Is there an Update coming for the Integration or should I download something manually?

Thx for your work and best regards.

Jo-Kaan, 2024-01-24:
Hello, 
I have installed/updated the new version via HACS. Unfortunately I can't notice any changes. The sensor values ​​continue to appear unavailable in the dark.

mawoka-myblock, 2024-01-24:
> Hello, I have installed/updated the new version via HACS. Unfortunately I can't notice any changes. The sensor values ​​continue to appear unavailable in the dark.

That's intended behavior: it's not a bug, it's a feature. If toh still think that the sensors should stay "available", feel free to reply, so we could consider changing it.

gerhardwallnoefer, 2024-01-25:
> That's intended behavior: it's not a bug, it's a feature. If toh still think that the sensors should stay "available", feel free to reply, so we could consider changing it.

That would be awesome. Thx for your work!

gerhardwallnoefer, 2024-01-25:
I also got these error in my logs:

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:898
First occurred: 08:39:37 (2 occurrences)
Last logged: 10:04:06

Update for sensor.apsystems_ez1_today_production fails
Update for sensor.apsystems_ez1_total_power fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 898, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1216, in async_device_update
    await self.async_update()
  File "/config/custom_components/apsystemsapi_local/sensor.py", line 226, in async_update
    await self.async_update_data()
  File "/config/custom_components/apsystemsapi_local/sensor.py", line 142, in async_update_data
    data = await self._api.get_output_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/APsystemsEZ1/__init__.py", line 163, in get_output_data
    return ReturnOutputData(**response["data"]) if response else None
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ReturnOutputData.__init__() missing 6 required positional arguments: 'p1', 'e1', 'te1', 'p2', 'e2', and 'te2'

gerhardwallnoefer, 2024-01-25:
And this error:

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:898
First occurred: 09:51:07 (1 occurrences)
Last logged: 09:51:07

Update for switch.apsystems_ez1_power_status fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 898, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1216, in async_device_update
    await self.async_update()
  File "/config/custom_components/apsystemsapi_local/switch.py", line 59, in async_update
    status = await self._api.get_device_power_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/APsystemsEZ1/__init__.py", line 251, in get_device_power_status
    return Status(int(response["data"]["status"])) if response else None
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: ''

gerhardwallnoefer, 2024-01-25:
And the last one:

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:1186
First occurred: 07:53:43 (11 occurrences)
Last logged: 07:53:43

Update of sensor.apsystems_ez1_today_production is taking over 10 seconds
Update of sensor.apsystems_ez1_today_production_p1 is taking over 10 seconds
Update of sensor.apsystems_ez1_today_production_p2 is taking over 10 seconds
Update of switch.apsystems_ez1_power_status is taking over 10 seconds
Update of number.apsystems_ez1_max_output_power is taking over 10 seconds

mawoka-myblock, 2024-01-29:
Sorry for the late response: All of that seems really strange, since the sensors should just become unavailable when the request fails, but your inverter responds some stuff, but really slow as well.

mawoka-myblock, 2024-03-20:
> > > Hello, I have installed/updated the new version via HACS. Unfortunately I can't notice any changes. The sensor values ​​continue to appear unavailable in the dark.
> > 
> > 
> > That's intended behavior: it's not a bug, it's a feature. If toh still think that the sensors should stay "available", feel free to reply, so we could consider changing it.
> 
> Hello, just wanted to ask if there is a chance that you update the code so that the sensor stay "available"? Thx

Out of interest, why?

mdouailin, 2024-03-23:
My 2 cents but I have an OLED display showing current prod / total prod and it's broken when I try to add an unavailable sensor, My use case is showing the sensor status (e.g. total daily prod) whatever the inverter status. It  seems legit for the inverter to go offline when there is no power, it's not the case for my Enphase envoy setup that consumes something about 4W when in the dark.

mawoka-myblock, 2024-03-29:
Okay, after doing some more research, I've found the following:

If we want to stick strictly to HA's guidelines, the sensor must become unavailable, as we can't communicate with the sensor anymore. See [here](https://developers.home-assistant.io/docs/core/entity#generic-properties) for HA's docs.

ccMatrix, 2024-04-04:
You can setup custom template sensors to handle these cases:

```
      apsystems_today_production:
        value_template: >
          {% if states('sensor.apsystems_ap_systems_ez1_today_production') == "unkown" %}
            {{ states('sensor.apsystems_today_production') }}
          {% elif states('sensor.apsystems_ap_systems_ez1_today_production') == "unavailable" %}
            {{ states('sensor.apsystems_today_production') }}
          {% elif states('sensor.apsystems_ap_systems_ez1_today_production') | float(0) > 0 %}
            {{ states('sensor.apsystems_ap_systems_ez1_today_production') }}
          {% else %}
            {{ states('sensor.apsystems_today_production') }}
          {% endif %}
        unit_of_measurement: 'kWh'
        device_class: energy
        friendly_name: 'APSystems Production Today'
      apsystems_total_production:
        value_template: >
          {% if states('sensor.apsystems_ap_systems_ez1_lifetime_production') == "unknown" %}
            {{ states('sensor.apsystems_total_production') }}
          {% elif states('sensor.apsystems_ap_systems_ez1_lifetime_production') == "unavailable" %}
            {{ states('sensor.apsystems_total_production') }}
          {% elif states('sensor.apsystems_ap_systems_ez1_lifetime_production') | float(0) > 0 %}
            {{ states('sensor.apsystems_ap_systems_ez1_lifetime_production') }}
          {% else %}
            {{ states('sensor.apsystems_total_production') }}
          {% endif %}
        unit_of_measurement: 'kWh'
        device_class: energy
        friendly_name: 'APSystems Production Total'
``` 
This way the value of these sensors will just stay the same even if the ap systems sensor becomes unavailable. The today production will also stay the same value until the next day when it starts producing again.

ccMatrix, 2024-04-04:
@gerhardwallnoefer the errors you posted are from the AP systems EZ1 API lib and how they handle responses from the API. I've opened an issue for it there: 

https://github.com/SonnenladenGmbH/APsystems-EZ1-API/issues/18

mawoka-myblock, 2024-04-04:
I just went through here and this issue is too cluttered now, as there are 2 points addressed in here:
1. Sensors go unavailable over night (Topic finished as this integration may be integrated into HA core, so sticking to their rules is a requirement -> setting unavailable is required)
2. The ValueError and TypeError mentioned by @gerhardwallnoefer. My take on that: Throwing these errors is acceptable as they are unexpected and (in my opinion) don't need to get handled in this regard.

I'm gonna hide the comments discussing point one as off-topic, hope that's fine for you.

mawoka-myblock, 2024-04-04:
Okay, I messed up. I created new issues to get these issues structured and organized again. Sorry for the mess right here. Above this comment, you can find the links to the corresponding issues.

Again, sorry for the inconvenience here as I haven't thought my previous comment through completely. As this doesn't help anyone anymore, I'll close this issue now, but feel free to add any additional details in the corresponding issues.

closed 2024-04-04 by mawoka-myblock