SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #15: Unable to prepare setup for platform apsystemsapi_local.sensor: Platform not found

Nelson666, 2024-02-24:
The HACS installation (2.0.1) are ok, but I have this error after a restart from HA. 

`Logger: homeassistant.setup
Source: setup.py:379
First occurred: 14:29:28 (3 occurrences)
Last logged: 14:29:28

Unable to prepare setup for platform apsystemsapi_local.sensor: Platform not found (cannot import name 'DeviceInfo' from 'homeassistant.helpers.device_registry' (/usr/src/homeassistant/homeassistant/helpers/device_registry.py)).
Unable to prepare setup for platform apsystemsapi_local.number: Platform not found (cannot import name 'DeviceInfo' from 'homeassistant.helpers.device_registry' (/usr/src/homeassistant/homeassistant/helpers/device_registry.py)).
Unable to prepare setup for platform apsystemsapi_local.switch: Platform not found (cannot import name 'DeviceInfo' from 'homeassistant.helpers.device_registry' (/usr/src/homeassistant/homeassistant/helpers/device_registry.py)).
`

mawoka-myblock, 2024-02-26:
Could you please tell me your HomeAssistant version?

Nelson666, 2024-02-28:
Sure, here are the versions. 

`Home Assistant 2023.8.3
Frontend 20230802.1 - latest `

mawoka-myblock, 2024-02-28:
Unfortunately I can't provide support for such an outdated version of Home Assistant. Please update HA and tell MSF of the issue persists.

Nelson666, 2024-02-28:
Ok, i update my HA installation and we will see. 

Thank you

closed 2024-03-29 by mawoka-myblock

hmueller01, 2024-04-26:
I have to use Home Assistant 2023.7.3 on my Synology nas, as 
https://github.com/hgy59/spksrc/tree/master/spk/homeassistant/src
doesn't support a newer version.

But if you replace 
```
-from homeassistant.helpers.device_registry import DeviceInfo
+from homeassistant.helpers.entity import DeviceInfo
```
in `sensor.py, switch.py, number.py` it works.

mawoka-myblock, 2024-04-26:
Keeping backwards compatibility isn't something I want to do, as the HA ecosystem seems to not care for that as well.

mawoka-myblock, 2024-04-26:
No problem, also thank you for providing the solution!