SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #18: EZ1-H

FlyingEagle, 2024-03-20:
Is this alspo usable for EZ1-___H____?

lostOzone, 2024-04-08:
It works but there is a problem with max_output_power it was 960 at first then I modified it and it can’t get it over 800 now.

lostOzone, 2024-04-21:
It doesn't work as the file modify is in an external API
https://github.com/SonnenladenGmbH/APsystems-EZ1-API
Do you kow how to midify it? I don't get a copy in /homeassistant/custom_components/

patman15, 2024-04-22:
Unless you are using the `core` installation method in which you will find the API in `HAcore/venv/lib/python3.11/site-packages/APsystemsEZ1` you want have access to this location. (Of course you can g[et root access to the HAOS](https://www.simon42.com/home-assistant-ssh-root-os/) and do it there.) Otherwise I think, I run out of ideas, especially simple ones.

mawoka-myblock, 2024-04-22:
Adding this to the library would be a lot of work, as it would have to check and save whether one has an H or M variant and adjust the limit accordingly.

lostOzone, 2024-04-22:
In the APsystems API getDeviceInfo return de real minPower and maxPower of the device not the configured one. So this should be implemented instead of hard coded values. I'll look into it but I never coded Python..

Scrat1608, 2024-05-23:
Bonjour.
Je  possède un EZ1-H, mais je ne parviens pas à l'intégrer.
J'ai une IP fixe.
Je suis en https pour me connecter à HA.
Je ne sais pas où est le problème.
Merci
-----------------------------
Good morning.
I have an EZ1-H, but I can't integrate it.
I have a fixed IP.
I'm in https to connect to Ha.
I don't know where the problem is.
THANKS

mawoka-myblock, 2024-05-23:
Please open an new issue for that and provide some more details please.

Scrat1608, 2024-05-23:
Ok. thanks