SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #19: problem to access az1

davjojo, 2024-03-23:
ping 192.168.1.90 is ok but log issue, could you help?

2024-03-23 11:56:29.288 WARNING (MainThread) [custom_components.apsystemsapi_local] Cannot connect to host 192.168.1.90:8050 ssl:default [Connect call failed ('192.168.1.90', 8050)]
2024-03-23 11:56:29.389 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=35 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=35, family=2, type=2, proto=0, laddr=('0.0.0.0', 49623)>

davjojo, 2024-03-23:
ping 192.168.1.90 is ok but log issue, could you help?

version Firmware: EZ1 1.7.1 (i cannot downgrad firmware)

Home assistant
Core
2024.3.3
Supervisor
2024.03.0
Operating System
12.1
Frontend
20240307.0

Réponse de 192.168.1.90 : octets=32 temps=139 ms TTL=255

Log
2024-03-23 11:56:29.288 WARNING (MainThread) [custom_components.apsystemsapi_local] Cannot connect to host 192.168.1.90:8050 ssl:default [Connect call failed ('192.168.1.90', 8050)]
2024-03-23 11:56:29.389 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=35 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=35, family=2, type=2, proto=0, laddr=('0.0.0.0', 49623)>

mawoka-myblock, 2024-03-23:
Are you sure that the local API is activated?

davjojo, 2024-03-23:
Enable newly added entities.
If newly discovered devices for APsystems Local API should be automatically added.
Enable polling for updates.
If Home Assistant should automatically poll APsystems Local API entities for updates.

is it what you want to know ?
yes 2 are enable

mawoka-myblock, 2024-03-23:
No, I wanted to ask if you enabled the local API in the APsystems app.

davjojo, 2024-03-23:
yes sorry i didn't to that.  its working very well now !
thank you very much.

last request : i only install HA for ap system, do you know if its possible to have it with domoticz ?
thank you

mawoka-myblock, 2024-03-23:
In theory yes, but someone would have to put their effort into that and one thing I know is that this person is not me, as I don't use Domoticz.

closed 2024-03-23 by mawoka-myblock

davjojo, 2024-03-23:
ok i understand, me i didn't know how to do.
thank