SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #22: Use of input.number for max Power

HABen91, 2024-03-26:
Is it possible to Switch the max Power setting from beeing a "number" to an "Input.number"?
This would allow to use a standard Automation to increase/decrease the value by increments. Doing this allows to realize a zero return  regulation to optimize power usage when using a Powerbank.

mawoka-myblock, 2024-03-26:
As far as I am aware, HA doesn't differentiate between these types. Could you please link some docs regarding that?

HABen91, 2024-03-26:
Hey,Here are some links to HA Dokumentation:https://www.home-assistant.io/integrations/number/https://www.home-assistant.io/integrations/input_number/In the attached Screenshot you will find the offered options to manipulate the different "kinds" of Numbers.

mawoka-myblock, 2024-03-26:
That's not how it works.
Here's a screenshot showing you how to achieve your goal:
[image]

closed 2024-03-26 by mawoka-myblock

HABen91, 2024-03-26:
The issue hast not been understood. The offered solution allows only fixed values but not incremantally reacting on other criteria.

mawoka-myblock, 2024-03-26:
Incrementing in watt-steps won't make sense as the inverter takes 1-2 seconds. Additionally, you seem to confuse the way HA integrations are built in contrast to helper integrations.
There are 3 solutions:
1. Use spook: https://spook.boo/number
2. Use templates to increment the value
3. Somebody spends time adding a custom service to this integration for this exact purpose and I won't be spending that time as I personally wouldn't benefit from that.