SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #23: Parallel work

patman15, 2024-03-30:
Hi!

I just saw that you pushed v3.0. I have been working on [similar changes](https://github.com/patman15/APsystems-EZ1-API-HomeAssistant) (central coordinator) plus some cleanups. This now somewhat redundant and I would need to rebase in order to figure out what the new deltas are and whether they make sense.
Can you have a look at my fork and let me know if you are interested in my changes / it makes sense for me to continue?
My bad that I did not ask before ...

mawoka-myblock, 2024-03-30:
Hey @patman15 , sorry that your work isn't needed anymore. You also missed one crucial thing: Handling when the inverter goes offline. I'm also preparing this integration to get merged into the core.

Again, I know that feeling and I feel sorry for you, but I still appreciate your work.

closed 2024-03-30 by mawoka-myblock

patman15, 2024-03-30:
@mawoka-myblock you might have missed, that I, among other things required, fixed sensor.py to better match current HA requirements to possibly get it upstreamed as discussed in https://github.com/SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant/issues/20. Nevertheless if you say you are good, fine for me. I might have a functional issue, but I will first test V3.

mawoka-myblock, 2024-03-30:
I'm in the process of creating the PR which is really hard. *takes a breath not to rant about HA*