SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #27: API thowing unhandled errors

mawoka-myblock, 2024-04-04:
Originally mentioned by @gerhardwallnoefer in #9.

The following exceptions were unhandled and experienced:

<details>
<summary>ValueError regarding an invalid integer</summary>

```
Update for switch.apsystems_ez1_power_status fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 898, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1216, in async_device_update
await self.async_update()
File "/config/custom_components/apsystemsapi_local/switch.py", line 59, in async_update
status = await self._api.get_device_power_status()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/APsystemsEZ1/init.py", line 251, in get_device_power_status
return Status(int(response["data"]["status"])) if response else None
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: ''
```

</details>

and

<details>
<summary>TypeError regarding missing inits for the ReturnOutputData</summary>

```
Update for sensor.apsystems_ez1_today_production fails
Update for sensor.apsystems_ez1_total_power fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 898, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1216, in async_device_update
await self.async_update()
File "/config/custom_components/apsystemsapi_local/sensor.py", line 226, in async_update
await self.async_update_data()
File "/config/custom_components/apsystemsapi_local/sensor.py", line 142, in async_update_data
data = await self._api.get_output_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/APsystemsEZ1/init.py", line 163, in get_output_data
return ReturnOutputData(**response["data"]) if response else None
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ReturnOutputData.init() missing 6 required positional arguments: 'p1', 'e1', 'te1', 'p2', 'e2', and 'te2'
```

</details>

My take on that: As these errors are so rare and are seemingly random, it's fine if they throw such an error.

skrue, 2024-05-22:
Hi,

I am also seeing the `ValueError` in my logs, and it's not exactly rare. However it appears in a pattern that I have not been able to reproduce: It starts appearing at a seemingly random point in time and then re-appears exactly every 5 minutes multiple times. Sometimes even for a period of about 1-2 hours. **But** occasionally a `InverterReturnedError` exception is raised after 5 minutes instead. So during the 1-2 hour period, I am seeing one of these two errors every 5 minutes.

Some more context: It seems that these errors pop up more often when the inverter is either under very light load (mornings, cloudy conditions) or under very heavy load for extended periods of time (afternoon on a very sunny day). My inverter is connected to four modules of 430Wp each (2 per string), so on sunny days I run into the output power limit pretty much all day long. My inverter is located in my basement very close to a WiFi access point, so connectivity should not be an issue and temperature-wise it should also be fine (at least it's not mounted up on the roof in the heat).

I think that given the frequency of these errors, they should be handled more gracefully.

Here's an example from my HA log:

```
2024-05-21 17:14:17.260 ERROR (MainThread) [homeassistant.helpers.entity] Update for switch.apsystems_ez1m_power_status fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 950, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1293, in async_device_update
await self.async_update()
File "/config/custom_components/apsystemsapi_local/switch.py", line 61, in async_update
status = await self._api.get_device_power_status()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 280, in get_device_power_status
return Status(int(response["data"]["status"])) if response else None
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: ''

...

2024-05-21 17:19:18.347 ERROR (MainThread) [homeassistant.helpers.entity] Update for switch.apsystems_ez1m_power_status fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 950, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1293, in async_device_update
await self.async_update()
File "/config/custom_components/apsystemsapi_local/switch.py", line 61, in async_update
status = await self._api.get_device_power_status()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 279, in get_device_power_status
response = await self._request("getOnOff")
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 97, in _request
return await self._request(endpoint, retry=False)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 98, in _request
raise InverterReturnedError
APsystemsEZ1.InverterReturnedError
```

skrue, 2024-05-22:
Okay, I'll have a log filter omit these errors then, so they don't clutter my logs... not very pretty, but as long as the integration keeps working I'm fine with it.