SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #30: Integration not loaded after launch.

DooMMasteR, 2024-04-08:
I have the feeling this is more related to recent changes on Home Assistant, but I am now often getting into the "Not loaded" situation.  
[image]
the Logs show no real error, which makes me suspect something Home Assistant side is not ready yet and prevents the integration from loading... restarting HA again often fixes it and once it has loaded it also survives nights and all.

mawoka-myblock, 2024-04-08:
Which HA version are you using?

DooMMasteR, 2024-04-24:
Current ones, I am guessing it's some race condition, but there are no errors in the HA logs.

wucke13, 2024-05-05:
I have the same behavior occuring since ~ a week: 
[image]

I'm on `2024.5.1`: 
[image]

DooMMasteR, 2024-05-06:
I can reproduce the issue better, when the inverter is offline, in that case the integration almost never loads.