SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #31: Inverter offline issue

patman15, 2024-04-08:
Hi!

Using v3.1.0, my logs are spammed with
```
2024-04-08 xx:xx:03.876 WARNING (MainThread) [homeassistant.components.number] Setup of number platform apsystemsapi_local is taking over 10 seconds.
2024-04-08 xx:xx:03.877 WARNING (MainThread) [homeassistant.components.switch] Setup of switch platform apsystemsapi_local is taking over 10 seconds.
2024-04-08 xx:xx:04.529 ERROR (MainThread) [homeassistant] Error doing job: Unclosed client session
2024-04-08 xx:xx:04.532 ERROR (MainThread) [homeassistant] Error doing job: Unclosed client session
```
Let me know if you need support fixing this. Since my last try to help, I'm a bit cautious in doing coding. :grin:

mawoka-myblock, 2024-04-08:
Hmmm... I thought I fixed it. Maybe you can point me to the error here?

https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/5c0c8be3d5ba59884f1f8d6e401547822aa620c0/APsystemsEZ1/__init__.py#L75-L93

patman15, 2024-04-08:
Having a quick guess I would say that https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/5c0c8be3d5ba59884f1f8d6e401547822aa620c0/APsystemsEZ1/__init__.py#L81 causes the session never to be closed.
Further, looking at https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/5c0c8be3d5ba59884f1f8d6e401547822aa620c0/APsystemsEZ1/__init__.py#L79 I would guess that you should use the `with` statement on the ClientSession so that it is automatically closed. Although, that is probably not what you want since you are differentiating between provided and generated session. Will think about it tomorrow if you still have the issue.

mawoka-myblock, 2024-04-08:
The problem is that HA can provide a ClientSession: https://github.com/home-assistant/core/pull/114531/files#diff-46b98840edeca8c80dc874ca885b2b84c476eef8b4bbf4fa21c661519fc7f50aR37-R39

That's how it's done in the PR.

I also don't want to break backwards compatibility.

mawoka-myblock, 2024-04-09:
Should be fixed in https://github.com/SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant/releases/tag/3.1.1