SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #32: Traceback with `APsystemsEZ1.InverterReturnedError` every 20 minutes with version 3.1.1 in HomeAssistant 2024.4.1, Docker compose setup

ngulden, 2024-04-10:
I see the following traceback in the logs of home assistant:

```
homeassistant  | 2024-04-10 10:06:21.867 ERROR (MainThread) [custom_components.apsystemsapi_local] Unexpected error fetching APSystems Data data
homeassistant  | Traceback (most recent call last):
homeassistant  |   File "/config/custom_components/apsystemsapi_local/__init__.py", line 103, in _async_refresh
homeassistant  |     self.data = await self._async_update_data()
homeassistant  |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/config/custom_components/apsystemsapi_local/__init__.py", line 77, in _async_update_data
homeassistant  |     data = await self.api.get_output_data()
homeassistant  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 191, in get_output_data
homeassistant  |     response = await self._request("getOutputData")
homeassistant  |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 97, in _request
homeassistant  |     return await self._request(endpoint, retry=False)
homeassistant  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 98, in _request
homeassistant  |     raise InverterReturnedError
homeassistant  | APsystemsEZ1.InverterReturnedError
```

I run Home Assistant 2024.4.1 in a Docker compose setup on a raspberry pi 4. I have version 3.1.1 of APSystems-EZ1-API-HomeAssistant through HACS installed. The integration works though. The traceback shows up in the logs every 20 minutes.

mawoka-myblock, 2024-04-10:
There's really nothing I can do about that, as the error gets thrown when the inverter responds with failed instead of success. But that's also everything we get to know from the inverter. Sorry, but this error doesn't originate from a bug in the code. Here are my suggestions on why the error appears:

- Little sun -> inverter reboots all the time
- Inverter broken
- Errors regarding voltage input

Sorry again!

closed 2024-04-10 by mawoka-myblock

ngulden, 2024-04-11:
Thanks for your response and the information.

However, I think the integration should handle the exception in a way that it doesn't throw a traceback to the logs. When you know, why it happens, you can handle it properly. This gives me as a user a better feeling, because I then know that the error can happen and the software handles it properly.

Leaving it as it is, is no good solution for me. Please reopen it and add a proper try..except block in your code.

BTW, my inverter works properly.