SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #33: Error when setting Max Output Power

dekoenpi, 2024-04-12:
APsystems Local API
3.1.1
Core
2024.4.2
Supervisor
2024.04.0
Operating System
12.2
Frontend
20240404.1

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2543, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2580, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 971, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1043, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 112, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/apsystemsapi_local/number.py", line 85, in async_set_native_value
    await self._api.set_max_power(int(value))
  File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 265, in set_max_power
    request = await self._request(f"setMaxPower?p={power_limit}")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 97, in _request
    return await self._request(endpoint, retry=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 98, in _request
    raise InverterReturnedError
APsystemsEZ1.InverterReturnedError
```

dekoenpi, 2024-04-12:
Follow-up:
All values below 601 work. Values from 601 - 800 do not.

mawoka-myblock, 2024-04-12:
Did you enable the 800w in the app?

dekoenpi, 2024-04-12:
Hi @mawoka-myblock 
Is this possible with "Direct Connection"? I can't find this in the "AP EasyPower" Mobile App.
Brgds

dekoenpi, 2024-04-15:
Ok, it only worked with an account and not in local mode. Thank you very much for your help.

closed 2024-04-15 by dekoenpi