SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #34: Restart of daily production

patman15, 2024-04-21:
Hi!

I'm having troubles with my statistics because the daily production counter restarts when the sun goes away and comes back. Mostly happens in the evening but also on very cloudy days. This causes further calculations that depend on that sensor to jump around. I would suggest to fix this by adding up the values of the inverter and reset only once at midnight. What do you think?
As previously, I can also suggest a fix (PR) if you want me to.

See the issue here:
[image]

mawoka-myblock, 2024-04-21:
Thanks for the offer, but I want this repo to get merged into HomeAssistant Core, so please don't open the PR here, but the PR is stuck at the moment, so I don't have a clue when this will get merged. You can check the PR here: https://github.com/home-assistant/core/pull/114531

patman15, 2024-04-22:
Ok understood, I'll do some local fixes and hope for the best for your PR :crossed_fingers:. Ping me if you want me to provide the fixes.

mawoka-myblock, 2024-04-22:
Once it's merged, you can submit the fixes directly to HA. But I'm not sure the PR will ever get merged.

patman15, 2024-04-28:
> But I'm not sure the PR will ever get merged.

I had a look at the code, are you sure that all prerequisites like the coding style are fulfilled? I had a quick look and my linters (ruff, pylint) show a number of issues.

mawoka-myblock, 2024-04-28:
Will check that. Could be that the merges have messed sth up. It's been some weeks since I've checked everything.

mawoka-myblock, 2024-04-28:
I've now fixed a new linter warning. The rules has been enabled in the last few weeks. Everything passes now.