SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #37: Log spam for platform switch and number

patman15, 2024-05-01:
Hi!

Sorry to bother you again, but I found this issue in version 3.1.1:
```
2024-04-xx 21:yy:14.463 WARNING (MainThread) [homeassistant.helpers.entity] Update of switch.apsystems_ez1_power_status is taking over 10 seconds
2024-04-xx 21:yy:14.464 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.apsystems_ez1_max_output_power is taking over 10 seconds
2024-04-xx 21:yy:44.464 WARNING (MainThread) [homeassistant.helpers.entity] Update of switch.apsystems_ez1_power_status is taking over 10 seconds
2024-04-xx 21:yy:44.465 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.apsystems_ez1_max_output_power is taking over 10 seconds
2024-04-xx 21:yy:14.466 WARNING (MainThread) [homeassistant.helpers.entity] Update of switch.apsystems_ez1_power_status is taking over 10 seconds
2024-04-xx 21:yy:14.466 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.apsystems_ez1_max_output_power is taking over 10 seconds
2024-04-xx 21:yy:44.468 WARNING (MainThread) [homeassistant.helpers.entity] Update of switch.apsystems_ez1_power_status is taking over 10 seconds
2024-04-xx 21:yy:44.469 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.apsystems_ez1_max_output_power is taking over 10 seconds
2024-04-xx 21:yy:14.469 WARNING (MainThread) [homeassistant.helpers.entity] Update of switch.apsystems_ez1_power_status is taking over 10 seconds
2024-04-xx 21:yy:14.470 WARNING (MainThread) [homeassistant.helpers.entity] Update of number.apsystems_ez1_max_output_power is taking over 10 seconds
```
Thanks a lot for taking the efforts to do this integration!

mawoka-myblock, 2024-05-01:
Thanks for mentioning this, but there's really nothing I can do about that. There's no logic in the integration that requires the time, so it's just the inverter. An alternative would be to set the timeout to eg. 9 seconds, but does that really make sense?

patman15, 2024-05-02:
Since you have the new PR, I would do a simple fix and change the timeout to 3 or 5 seconds. It's quite obvious that this is the timeout of the asyncio. I also consider it very unlikely, that the inverter really takes more than 3 or 5 seconds to answer a HTTP request on a WLAN ...

mawoka-myblock, 2024-05-02:
Okay, I'll do that. Thank you for pointing this out.

mawoka-myblock, 2024-05-05:
Oh, as the PR only adds the sensor platform, that's irrelevant for that.