SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #43: Issue with EZ1 integration and Shelly 3EM

janfullert, 2024-05-17:
Hello,
at first thanks for your great work on it. I love to implement my EZ1 to homeassistant
I got a problem when I integrate the Solar-Production in my energy-dashboard.
Until this, the energy-dashboard measure my energy consumption correct. 
I am measuring my energy with an Shelly 3EM. The energy input of the solar-production get in on phase C and is directly subtract there an reducing my grid input.
After implementing the Solar-System the energy I consume myself is sum up on this and increase the whole energy consumption.
In that case the solar-energy which I consume myself is count twice. 
I already tried to config the energy-dashboard by using sensors or not but it doesn't work.
Do you got an Idea for solving this for me.

mawoka-myblock, 2024-05-17:
Could you provide us with a screenshot of your dashboard config?

janfullert, 2024-05-17:
[image]
[image]

as you see in the second one phase C is 0,48kWh - this is already including my self consumption - and at the end everything is thum up and the returning is subtracted again
I also added a Screenshot of my Shelly-Dashboard

[image]

mawoka-myblock, 2024-05-17:
I am still struggling to understand the issue. The card in the top right (the flow one) is correct, right? But the one below telling you that you've consumed 0,8kwh isn't, right?

Btw: interesting google searches going on there ;)

janfullert, 2024-05-17:
:-)) yes google is my helping hand - I am already just a beginner

Yes the pruduction with 3kWH is correct. The down left its correct too. From grid I consumed 2,7kWh an 1,9kWh returned. The one down right is wrong. I consumed 2,7kWh from grid + 0,9kW from Solar.

I hope you understand - my english is not the best...

janfullert, 2024-05-17:
Sorry - I consumed 2,7kWh + 1,1kWh -> the difference between production 3kWh an returning 1,9kWh

mawoka-myblock, 2024-05-20:
Sorry for the late reply, but I think that this may be a question for the HA forum, as this doesn't seem to be directly linked to this integration. Hope you'll find a solution, but unfortunately your issue goes over my head.