SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant #48: Connection EZ1-H not possible.

Scrat1608, 2024-05-23:
Bonjour.
Je possède un EZ1-H, mais je ne parviens pas à l'intégrer.
J'ai une IP fixe.
Je suis en https pour me connecter à HA.
Je ne sais pas où est le problème.
Merci
---------------------
Good morning.
I have an EZ1-H, but I can't integrate it.
I have a fixed IP.
I'm in https to connect to Ha.
I don't know where the problem is.
THANKS

Scrat1608, 2024-05-23:
Je ne sais pas quels détails supplémentaire vous pouvez avoir besoin.
Est ce que le fait Je ne sais pas quels détails supplémentaire vous pouvez avoir besoin.
--
I don't know what additional details you may need.
Is the fact that I am in HTTP prevents the discovery of the IP address?

mawoka-myblock, 2024-05-23:
No, that's not the problem. Is the local api enabled in the app?

Scrat1608, 2024-05-23:
J'ai activé l'ip local directement dans la box.
Dans l'application je n'ai pas la possibilité de mettre une ip local.
---
I activated the local IP directly in the box.
In the application I do not have the possibility of putting a local IP.

Scrat1608, 2024-05-23:
Il y a peut être un port à ouvrir ?
--
Maybe there is a port to open?

Scrat1608, 2024-05-23:
Peut-être que je n'ai pas fixé la bonne IP.
Comment connaitre l’adresse mac de l'onduleur ?
--
Maybe I haven't set the right IP.
How to know the Mac address of the inverter?

Scrat1608, 2024-05-24:
J'avais un probleme d'ip locale qui n'était pas bon. Merci. sujet clos.

closed 2024-05-24 by Scrat1608